PaulStoffregen/Wire #26: Nouveau composant en @I2C a rajouter dans l'exemple "Scanner" de l'IDE ARDUINO

AIRPOPCIE, 2021-02-07:
Bonjour, j'ai un multiplexeur TCA9548A dont @ est 0x70h.
case 0x70: Serial.print(F("HT16K33,TCA9548A")); break;
[Scanner_I2C.zip](https://github.com/PaulStoffregen/Wire/files/5938548/Scanner_I2C.zip)

PaulStoffregen, 2021-02-07:
Added https://github.com/PaulStoffregen/Wire/commit/18d892e454c102482f1de92dc518896f7d536f1e

closed 2021-02-07 by PaulStoffregen

AIRPOPCIE, 2021-02-08:
en fait le TCA9548A peut se programmer sur 8 octets. 0x70H à 0x77H.
Modification de l'adresse I2C du TCA9548A
L'adresse de bus du TCA9548A est modifiée à l'aide des connexions aux broches A0, A1 et A2. Par défaut dans le tutoriel, nous utilisons 0x70, en câblant A0 ~ A2 à GND (connu sous le nom de LOW). En utilisant le tableau ci-dessous, vous pouvez reconfigurer à une adresse entre 0x70 et 0x77 en faisant correspondre les entrées à HIGH (5V) ou LOW (GND):
donc on obtient:

case 0x70:   Serial.print(F("HT16K33,**TCA9548A**")); break;
case 0x71:   Serial.print(F("SFE7SEG,HT16K33,**TCA9548A**")); break;
case 0x72:   Serial.print(F("HT16K33,**TCA9548A**")); break;
case 0x73:   Serial.print(F("HT16K33,**TCA9548A**")); break;
case 0x76:   Serial.print(F("MS5607,MS5611,MS5637,BMP280,**TCA9548A**")); break;
case 0x77:   Serial.print(F("BMP085,BMA180,BMP280,MS5611,**TCA9548A**")); break;

")); break;