PaulStoffregen/Wire #42: Teensy 4.1 - I2C bus is locked if trying to read not connected sensor

mateusz-kusmierz, 2023-05-13:
Hi.
In my project I've encountered an issue. I have 3 temperature sensors LM75A. Normally they are all connected and everything works.
But if one of them is not connected, I2C bus is locked (no traffic at all) - it looks like it is waiting for unconnected sensor to response but that never happens.

I thought there is setWireTimeout but couldn't find that in this library.

How do I set timeout in this library, or how do I solve this issue other than "just connect all 3 sensors"?

PaulStoffregen, 2023-05-13:
If you run the Scanner example, you can see it tries to communicate with every address and prints those which do respond.  Obviously it is not locking up with the addresses where no I2C device exists.

Please post a small but complete program which demonstrates how to reproduce this lockup, but just running on a Teensy 4.1 without any LM35 chips connected.

mateusz-kusmierz, 2023-05-16:
Hi. Sorry Paul. I've done some investigation and this simple example reproduces "locking" behaviour that I'm having in my project:

```cpp
#include <Temperature_LM75_Derived.h>
#include <EventResponder.h>

EventResponder event;
Generic_LM75 lm75a(0x4A);    //mainboard

void temperaturePrint() {
  Serial.print("Temperature = ");
  Serial.println(lm75a.readTemperatureC());
}

void eventFn(EventResponderRef r) {
  temperaturePrint();
  delay(1000);
  event.triggerEvent();
}

void setup() {
  Serial.begin(115200);

  Wire.begin();

  temperaturePrint();
  event.attach(&eventFn);
  event.triggerEvent();
  temperaturePrint();
}

void loop() 
{
}

```
Serial Monitor output:
```
Temperature = 23.00
Temperature = Temperature = 23.00
Temperature = -0.50
Temperature = -0.50
-0.50
Temperature = -0.50
Temperature = -0.50
Temperature = -0.50
Temperature = -0.50
Temperature = -0.50
Temperature = -0.50
Temperature = -0.50
Temperature = -0.50
Temperature = -0.50
Temperature = -0.50
Temperature = -0.50
Temperature = -0.50
Temperature = -0.50
Temperature = -0.50
Temperature = -0.50
Temperature = -0.50
Temperature = -0.50
Temperature = -0.50
Temperature = -0.50
Temperature = -0.50
Temperature = -0.50
```
Sensor was connected all the time.